TheCommsChannel/TC2-BBS-mesh #3: Not getting a reply

ronnietucker, 2024-06-27:
Hi!

I've got everything installed on my Linux laptop (Mint). I've got the BBS connected to my router/client (device1) that's on my roof (it's using MQTT if that makes any difference?) by using TCP and its IP address in the config file.

Got the BBS running in the terminal. No errors there.

When I send from device2 to device1 I get no reply.

Any ideas?

**PS**: additional setup step in Linux was to run 
`sudo apt install python3.10-venv `

TheCommsChannel, 2024-06-27:
What type of device are you connecting to via TCP?

ronnietucker, 2024-06-27:
> What type of device are you connecting to via TCP?

It's a Heltec V3 with the latest firmware. Both my laptop and the Heltec (device1) are on the same network and the BBS config has the IP address of the Heltec unit. The BBS software seems to be running fine and sitting waiting for a message. The message reaches device1 but it never seems to register with the BBS software.

The device2 is also a Heltec V3. I'm connecting to that with my phone and the Meshtastic app.

ronnietucker, 2024-06-28:
Forget this. There seems to be an issue when the BBS thing crashes and even after a restart of the server.py it doesn't recognise anything.

A full system reboot fixes it. I'll file a separate bug report.

closed 2024-06-28 by ronnietucker