TheCommsChannel/TC2-BBS-mesh #4: server errors

gadd1, 2024-06-27:
Hi
First, great initiative and thanks for working on that. To the most part between the nodes I have at home its working OK.

I did notice several errors prompts. not sure if they relate 

first one:
2024-06-27 19:33:28,272 - ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>
2024-06-27 19:34:23,247 - ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>

second type of error:
Traceback (most recent call last):
  File "/home/.local/lib/python3.11/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/.local/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/.local/lib/python3.11/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/.local/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/home/.local/lib/python3.11/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface

Running on Raspberry Pi Zero 2

KE6MTO, 2024-06-27:
I got the same error on a Ubuntu VM. Stopping/Starting the server.py cleared it up for now. Not sure if it was a command issue or a packet it didn't understand. The node that is connected to the server is running 2.3.14.

TheCommsChannel, 2024-06-27:
I believe this is related to a packet it doesn't understand as @KE6MTO suggests. I'm able to run without errors in my test environment, but switching to LongFast where everyone else is causes this error from time to time. I've not been able to track it down yet.

The error seems to not affect the operation of the script luckily.

KE6MTO, 2024-06-27:
On mine it would still receive packets (and display on the console) but it would stop responding until I restarted the server.

gadd1, 2024-06-27:
wow, thanks for the fast response guys
I suspect the issue relate the the python library on the Pi OS
it was running from the /bin/python3 and when switching to the /usr/bin/python3 it seems to work now
/usr/bin/python3 server.py

ianmcorvidae, 2024-06-27:
The one to do with the interface I believe you can probably fix by accepting an `interface` argument on the function you're registering as the pubsub callback for `meshtastic.receive`. Won't do that much other than silencing the error, probably, but could still be worth it.

ronnietucker, 2024-06-29:
Yeah, I'm also getting the same error when the unit (Heltec V3) is sitting doing nothing.

```
2024-06-29 14:15:30,630 - ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>
Traceback (most recent call last):
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface
```

ianmcorvidae, 2024-07-02:
Linked a couple issue to this that include similar stuff. The fix for `SenderUnknownMsgDataError' is in https://github.com/TheCommsChannel/TC2-BBS-mesh/issues/24#issuecomment-2198330622

ronnietucker, 2024-07-02:
> Linked a couple issue to this that include similar stuff. The fix for `SenderUnknownMsgDataError' is in [#24 (comment)](https://github.com/TheCommsChannel/TC2-BBS-mesh/issues/24#issuecomment-2198330622)

Does that fix it though, or just silence the error?

ianmcorvidae, 2024-07-02:
It fixes it, though the error is ultimately benign (it's just complaining that it's being given more information that it expects to be, basically). Making the change does make the code slightly more correct, though, in that it's then using the interface passed along with the packet, rather than the one defined within server.py -- of course, at present those are the same interface :)

TheCommsChannel, 2024-07-02:
Fix seems to be working and just updated the code with the it. Thanks everyone!

https://github.com/TheCommsChannel/TC2-BBS-mesh/commit/be9aa375cf0ed8f96fb2a51e37f87698ff26d36f

closed 2024-07-02 by TheCommsChannel