TheCommsChannel/TC2-BBS-mesh #11: Run from boot

se7en9057, 2024-06-28:
How would someone get this to run from boot? i'm having to start the server by running source venv/bin/activate then python server.py to get things running. but it doesn't seem to be running as a service. is there a way to either run this as a service or have it start when the device starts?

upchui, 2024-06-28:
You could probably create a systemd service for this.

TheCommsChannel, 2024-06-28:
systemd service as @upchui suggests is the way to go. I'll add this info to the README tomorrow, but you can see an example on how this is done in the bottom of the README for my SkyWatch script if you want to follow those steps

https://github.com/TheCommsChannel/SkyWatch

sabotrax, 2024-06-28:
Another way is tmux, but it's not as capable as systemd.

- Install tmux
- In the home directory of the user running the server create ".tmux.conf":
`new-session -d -s TC2BBS`
`send-keys -t TC2BBS "/bin/bash" C-m`
`send-keys -t TC2BBS "cd ~/TC2-BBS-mesh" C-m`
`send-keys -t TC2BBS "source venv/bin/activate" C-m`
`send-keys -t TC2BBS "python3 server.py" C-m`
- Add to the crontab of the same user:
`@reboot /usr/bin/tmux`

The server will run in tmux after reboot.
If you want to check its output, attach the tmux session to your terminal with `tmux a`. Detach with `ctrl + b` than `d`.
tmux [cheat sheet](https://tmuxcheatsheet.com/).

This is working on Debian 12. Your paths or shell may differ.

TheCommsChannel, 2024-06-28:
Added service file and instructions on how to set this up.

https://github.com/TheCommsChannel/TC2-BBS-mesh?tab=readme-ov-file#automatically-run-at-boot

closed 2024-06-28 by TheCommsChannel