TheCommsChannel/TC2-BBS-mesh #18: How to completely restart the BBS?

ronnietucker, 2024-06-28:
I've had it working then it'll crash with an error. When I restart the BBS it won't receive anything.

I've been restarting it (on Linux) with a CTRL+C then run the server.py command again. No dice though.

What's the best way to completely restart the BBS?

**Example crash**:

```
2024-06-28 18:07:06,717 - INFO - Received message from user 'RTj2' to RTj1: Bulletin test
2024-06-28 18:07:40,058 - INFO - Received message from user 'RTj2' to RTj1: END
2024-06-28 18:07:50,280 - INFO - Received message from user 'RTj2' to RTj1: 2
2024-06-28 18:07:57,863 - INFO - Received message from user 'RTj2' to RTj1: This is a test of the TC2 bulletin feature
2024-06-28 18:07:57,868 - ERROR - Unexpected error in deferred execution <class 'ValueError'>
Traceback (most recent call last):
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/ronnie/TC2-BBS-mesh/server.py", line 48, in receive_packet
    on_receive(packet, interface)
  File "/home/ronnie/TC2-BBS-mesh/message_processing.py", line 92, in on_receive
    process_message(sender_id, message_string, interface, is_sync_message=False)
  File "/home/ronnie/TC2-BBS-mesh/message_processing.py", line 60, in process_message
    handle_bb_steps(sender_id, message, step, state, interface, bbs_nodes)
  File "/home/ronnie/TC2-BBS-mesh/command_handlers.py", line 159, in handle_bb_steps
    board_name = boards.get(int(message))
ValueError: invalid literal for int() with base 10: 'This is a test of the TC2 bulletin feature'

```

ronnietucker, 2024-06-28:
Another example crash:
(I did a CTRL+C after the OSError message)

```
2024-06-28 18:35:13,399 - INFO - TC²-BBS is running on tcp interface...
Database schema initialized.
2024-06-28 18:35:13,412 - ERROR - Unexpected OSError, terminating meshtastic reader... [Errno 104] Connection reset by peer
^C2024-06-28 18:35:21,093 - INFO - Shutting down the server...
Traceback (most recent call last):
  File "/home/ronnie/TC2-BBS-mesh/server.py", line 53, in main
    while True:
KeyboardInterrupt

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ronnie/TC2-BBS-mesh/server.py", line 60, in <module>
    main()
  File "/home/ronnie/TC2-BBS-mesh/server.py", line 57, in main
    interface.close()
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/tcp_interface.py", line 60, in close
    StreamInterface.close(self)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/stream_interface.py", line 120, in close
    MeshInterface.close(self)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 122, in close
    self._sendDisconnect()
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 758, in _sendDisconnect
    self._sendToRadio(m)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 783, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
```

TheCommsChannel, 2024-06-28:
Are you able to try with a USB connected device? I wonder if the issues are TCP related.

ronnietucker, 2024-06-29:
I was using TCP in the above message, but here's some errors I got while using a USB connection in Linux.

```
2024-06-29 12:25:16,160 - INFO - Received message from user 'RTj1' to RTj2: 0
2024-06-29 12:25:16,176 - ERROR - Unexpected error in deferred execution <class 'TypeError'>
Traceback (most recent call last):
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/ronnie/TC2-BBS-mesh/server.py", line 48, in receive_packet
    on_receive(packet, interface)
  File "/home/ronnie/TC2-BBS-mesh/message_processing.py", line 92, in on_receive
    process_message(sender_id, message_string, interface, is_sync_message=False)
  File "/home/ronnie/TC2-BBS-mesh/message_processing.py", line 58, in process_message
    handle_mail_steps(sender_id, message, step, state, interface, bbs_nodes)
  File "/home/ronnie/TC2-BBS-mesh/command_handlers.py", line 263, in handle_mail_steps
    sender, date, subject, content, unique_id = get_mail_content(mail_id)
TypeError: cannot unpack non-iterable NoneType object

2024-06-29 12:25:31,670 - INFO - Received message from user 'RTj1' to RTj2: M
```

With this one (below) I was in the middle of making a message when I was getting no reply after the subject.

```
2024-06-29 12:27:42,064 - INFO - Received message from user 'RTj1' to RTj2: M
2024-06-29 12:27:49,586 - INFO - Received message from user 'RTj1' to RTj2: 1
2024-06-29 12:28:05,627 - INFO - Received message from user 'RTj1' to RTj2: RTj3
2024-06-29 12:28:29,493 - INFO - Received message from user 'RTj1' to RTj2: Test from RTj1
2024-06-29 12:28:57,276 - INFO - Received message from user 'RTj1' to RTj2: Test message
2024-06-29 12:29:14,972 - INFO - Received message from user 'RTj1' to RTj2: Hello?
2024-06-29 12:29:23,937 - INFO - Received message from user 'RTj1' to RTj2: M
2024-06-29 12:29:31,197 - INFO - Received message from user 'RTj1' to RTj2: 1
2024-06-29 12:29:39,734 - INFO - Received message from user 'RTj1' to RTj2: RTj3
2024-06-29 12:29:49,339 - INFO - Received message from user 'RTj1' to RTj2: Test message
2024-06-29 12:30:08,618 - INFO - Received message from user 'RTj1' to RTj2: Subject line
2024-06-29 12:30:15,387 - ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>
Traceback (most recent call last):
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface

2024-06-29 12:30:27,990 - ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>
Traceback (most recent call last):
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface

2024-06-29 12:30:31,532 - INFO - Received message from user 'RTj1' to RTj2: 2
2024-06-29 12:30:38,215 - ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>
Traceback (most recent call last):
[... cut ...]
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface

2024-06-29 12:30:47,803 - INFO - Received message from user 'RTj1' to RTj2: M
```