TheCommsChannel/TC2-BBS-mesh #20: suggestion: Add recommended hardware to readme

joeketcham, 2024-06-28:
So far I've tried the following combinations: 

RPI3 with Rak starter kit (serial connection)
  - It only stays alive for approximately 40 minutes then the server cannot read traffic from the radio anymore

Windows 10 with Rak starter kit (serial connection)
  - It only stays alive for approximately 40 minutes then the server cannot read traffic from the radio anymore

Windows 10 with TBeam ESP32 chipset (serial connection)
  - Stable 

I'll try the TBeam with RPI3 later tonight and see if it's also stable. I figure the RAK being an ultra low power device must not maintain a stable connection over USB and cycles occasionally to conserve power while the TBeam does not. Though in your videos it looked like you were using a RAK so I'm not sure. 

Can others confirm similar results or state if their RAKs work?

joeketcham, 2024-06-28:
I do have extra RAKs, I will try another RAK to see if it was just the one radio or if the problem follows with other RAKs too. Curious what other people are seeing with this

TheCommsChannel, 2024-06-29:
May be something with your RAK. BBS1 shown on the video is a RAK with Pi Zero (version 1). Been running for a few days now with no issue. BBS2 is a Heltec V3 connected to the uConsole and BBS3 is another RAK connected to a Linux Mint VM.

OHTechDiver, 2024-06-29:
I've had similar issues with various hardware. I've only been able to successfully use a combination of Raspberry Pi 4 (Bookworm 64-bit) with RAK4631 (12.3.14) for about 30-40 minutes before server stops reading traffic.  Fresh install fails to communicate with ESP32-based devices (Heltec V3 and T-Beam Supreme) via serial connection.

joeketcham, 2024-06-29:
Solution: I figured it out, at least for my hardware perhaps OHTechDiver can test same on his: 

RAK was configured as a 'router'. I was using this as a repeater/router (repeater doesn't report battery status) out in the field and when I pulled it to use with the BBS I never gave it a thought. After 30-40 minutes as a router the BBS could no longer read traffic. The router would still receive and repeat, but BBS couldn't see it or respond. 

I switched the RAK to 'client' and boom, off to the races. It's been stable now for over 4 hours no issue.

TheCommsChannel, 2024-06-29:
Nice catch! Mine were all in 'client' as well during testing so this makes sense. The nodes I intend to deploy out in the field are router_client though 😬 I wonder if that role is affected as well.

joeketcham, 2024-06-29:
I submitted a PR to update the readme with a note regarding device role to reflect what I found in testing

thebigshed, 2024-06-29:
Mine is set to CLIENT but still get the errors on RPI Rasbian 32bit and on Ubuntu 64 bit. Just enabled the serial module on the LILYGO and now am trying the different protocols. Started with Default. What do you use?

joeketcham, 2024-06-29:
Only errors I saw were low power warnings. I had to play around with bigger power supplies and higher quality cables. Otherwise I haven't seen any more errors. Our testing as included: 
- Sent mail between users
- Posted general bulletins 
- Urgent bulletin sends the subject to the main chat
- Fortune works
- Wall of shame is reporting users with low battery 

I haven't tried the sync from BBS to BBS yet, but now that we have two running stable I'm sure we'll be doing that soon too

TheCommsChannel, 2024-06-29:
Testing out router_client now. I'll give it until later tonight and I'll add it to the README if it still works

adam-burns, 2024-06-30:
> Nice catch! Mine were all in 'client' as well during testing so this makes sense. The nodes I intend to deploy out in the field are router_client though 😬 I wonder if that role is affected as well.

I've been running T-ECHO in router-client mode via USB to RPi4 (OS: Devuan Daedalus) for a day or so & seems to be working.

joeketcham, 2024-07-09:
I think this is safe to close at this point. We noted the most important setting in the readme and discovered why the BBS would stop talking to the radio after 40-60 minutes if the role is miss-configured.

closed 2024-07-09 by joeketcham