TheCommsChannel/TC2-BBS-mesh #23: errors

out0fsync, 2024-06-28:
Not sure if any of this is useful to you. Great idea :) Running on orangepi zero, ubuntu:-

2024-06-28 21:57:42,307 - INFO - TC²-BBS is running on tcp interface...
Database schema initialized.
2024-06-28 21:58:19,573 - INFO - Received message from user 'd109' to c300: hello
2024-06-28 22:02:16,959 - INFO - Received message from user 'd109' to None: the bbs is running on my c300 node just direct message "hello" to it
2024-06-28 22:02:16,960 - INFO - Ignoring message sent to group chat or from unknown node
2024-06-28 22:04:25,114 - INFO - Received message from user 'd109' to c300: B
2024-06-28 22:05:05,097 - INFO - Received message from user 'd109' to c300: [b]
2024-06-28 22:05:05,098 - ERROR - Unexpected error in deferred execution <class  'ValueError'>
Traceback (most recent call last):
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/util.py"   , line 283, in _run   o()
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/root/TC2-BBS-mesh/server.py", line 49, in receive_packet on_receive(packet, interface)
  File "/root/TC2-BBS-mesh/message_processing.py", line 92, in on_receive
    process_message(sender_id, message_string, interface, is_sync_message=False)
  File "/root/TC2-BBS-mesh/message_processing.py", line 60, in process_message
    handle_bb_steps(sender_id, message, step, state, interface, bbs_nodes)
  File "/root/TC2-BBS-mesh/command_handlers.py", line 159, in handle_bb_steps
    board_name = boards.get(int(message))
ValueError: invalid literal for int() with base 10: '[b]'

2024-06-28 22:05:48,360 - INFO - Received message from user 'd109' to c300: B
2024-06-28 22:30:20,373 - ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>
Traceback (most recent call last):
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface

2024-06-28 22:40:22,223 - INFO - Received message from user 'd109' to c300: [1]
2024-06-28 22:40:22,224 - ERROR - Unexpected error in deferred execution <class 'ValueError'>
Traceback (most recent call last):
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/root/TC2-BBS-mesh/server.py", line 49, in receive_packet
    on_receive(packet, interface)
  File "/root/TC2-BBS-mesh/message_processing.py", line 92, in on_receive
    process_message(sender_id, message_string, interface, is_sync_message=False)
  File "/root/TC2-BBS-mesh/message_processing.py", line 60, in process_message
    handle_bb_steps(sender_id, message, step, state, interface, bbs_nodes)
  File "/root/TC2-BBS-mesh/command_handlers.py", line 159, in handle_bb_steps
    board_name = boards.get(int(message))
ValueError: invalid literal for int() with base 10: '[1]'
[... 50 lines cut ...]
  File "/root/TC2-BBS-mesh/command_handlers.py", line 159, in handle_bb_steps
    board_name = boards.get(int(message))
ValueError: invalid literal for int() with base 10: '[2]'

2024-06-28 23:04:47,126 - INFO - Received message from user 'd109' to c300: 2
2024-06-28 23:05:01,458 - INFO - Received message from user 'd109' to c300: 0
2024-06-28 23:05:22,809 - INFO - Received message from user 'd109' to c300: hello
2024-06-28 23:33:36,974 - INFO - Received message from user 'd109' to None: hello
2024-06-28 23:33:36,975 - INFO - Ignoring message sent to group chat or from unknown node
2024-06-28 23:34:37,176 - INFO - Received message from user 'd109' to c300: 2
2024-06-28 23:34:49,965 - INFO - Received message from user 'd109' to c300: 4
2024-06-28 23:35:03,267 - INFO - Received message from user 'd109' to c300: f
2024-06-29 00:04:26,331 - ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>
Traceback (most recent call last):
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface

thebigshed, 2024-06-29:
**getting similar on RPI on serial connection:**

2024-06-29 16:30:36,769 - ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>
Traceback (most recent call last):
  File "/home/david/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/david/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/home/david/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/david/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/home/david/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface

**using Docker to run the image.**

ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>

out0fsync, 2024-06-30:
now getting this error whenever I try and run BBS

2024-06-30 23:13:14,059 - INFO - TC²-BBS is running on tcp interface...
Database schema initialized.
2024-06-30 23:13:14,115 - ERROR - Unexpected OSError, terminating meshtastic reader... [Errno 104] Connection reset by peer

ianmcorvidae, 2024-07-02:
Just linking identical issues together: The `SenderUnknownMsgDataError` here is the same reported in #4 and which is fixed by the change in comment: https://github.com/TheCommsChannel/TC2-BBS-mesh/issues/24#issuecomment-2198330622

ValueError and OSError issues probably separate.

TheCommsChannel, 2024-07-02:
The errors from the original post have been fixed in the latest update so I'll go ahead and close this. If you're still receiving the errors with ValueError and OSError, please open up a new issue.

closed 2024-07-02 by TheCommsChannel

out0fsync, 2024-07-02:
How do I update? Is it just a case of running pip install requirements
again?