TheCommsChannel/TC2-BBS-mesh #24: No Response on TCP connection.

ItsNix, 2024-06-29:
Running on Docker with a node attached to wifi.

I can see the incoming messages but do not get a response on my Meshtastic node.

2024-06-29 08:45:39,997 - INFO - TC²-BBS is running on tcp interface...
2024-06-29 08:46:17,076 - INFO - Received message from user 'AsMo' to MDev: Hello
2024-06-29 08:47:06,939 - INFO - Received message from user 'AsMo' to MDev: ?
2024-06-29 08:48:21,796 - INFO - TC²-BBS is running on tcp interface...
2024-06-29 08:48:36,358 - INFO - Received message from user 'AsMo' to MDev: Check
2024-06-29 08:53:14,803 - INFO - Received message from user 'AsMo' to MDev: 1
2024-06-29 08:55:24,414 - INFO - Received message from user 'AsMo' to MDev: m
2024-06-29 08:55:31,128 - INFO - Received message from user 'AsMo' to MDev: M
2024-06-29 08:59:32,264 - INFO - Received message from user 'AsMo' to MDev: 0
2024-06-29 09:00:27,374 - INFO - Received message from user 'AsMo' to MDev: h
2024-06-29 09:00:37,591 - INFO - Received message from user 'AsMo' to MDev: H
2024-06-29 09:04:27,300 - INFO - Received message from user 'AsMo' to MDev: F
2024-06-29 09:04:34,586 - INFO - Received message from user 'AsMo' to MDev: W

I will keep tinkering but any guidance is welcomed.

TheCommsChannel, 2024-06-29:
I've not done much testing with TCP, but seems like many others are having issues as well. Recommend serial only for now.

ronnietucker, 2024-06-29:
> I've not done much testing with TCP, but seems like many others are having issues as well. Recommend serial only for now.

I'm going to leave TCP running on mine for today and see if I can get others nearby to give it a test run.

Craig-Hz, 2024-06-29:
I had success using Serial via TCP with a T3S3 but no luck using TCP with a Station G2 Nano.

Phlat, 2024-06-29:
I'm using a WaveShare LoraWan Hat on a Raspberry Pi running Meshtasticd. The BBS is mostly working over TCP but of course there is not a Main help menu response to a direct message and only some of the commands work.  I get Meshtastic API interface errors. It may have something to do with the pubsub topic. `Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>`

typicalaimster, 2024-06-29:
@Phlat try changing def receive_packet(packet) to def receive_packet(packet, interface) in server.py. Seems like a bug there.

Phlat, 2024-06-29:
Interesting that I no longer get that specific error however it does not fix the TCP issue. Still no reply to hello. 

I switched to a Heltec Lora32 on the usb port, changed configs. It does produce the hello menu as before, however there are more bugs. The F fortune always works however sending a Hello throws an error:
2024-06-29 14:28:27,889 - INFO - Received message from user '4TWO' to YBBS: F
2024-06-29 14:29:12,296 - INFO - Received message from user '4TWO' to YBBS: Hello
2024-06-29 14:29:12,297 - ERROR - Unexpected error in deferred execution <class 'ValueError'>
Traceback (most recent call last):
  File "/home/cms/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/cms/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/cms/TC2-BBS-mesh/venv/lib/python3.11/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/cms/TC2-BBS-mesh/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/cms/TC2-BBS-mesh/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/cms/TC2-BBS-mesh/venv/lib/python3.11/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/cms/TC2-BBS-mesh/server.py", line 61, in receive_packet
    on_receive(packet, interface)
  File "/home/cms/TC2-BBS-mesh/message_processing.py", line 92, in on_receive
    process_message(sender_id, message_string, interface, is_sync_message=False)
  File "/home/cms/TC2-BBS-mesh/message_processing.py", line 62, in process_message
    handle_stats_steps(sender_id, message, step, interface, bbs_nodes)
  File "/home/cms/TC2-BBS-mesh/command_handlers.py", line 108, in handle_stats_steps
    choice = int(choice)
             ^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'HELLO'

2024-06-29 14:29:53,455 - INFO - Received message from user '4TWO' to YBBS: F`

typicalaimster, 2024-06-29:
Interesting as I've had no problems TCP to both a T-Beam, Portdruno, and Radio Master.

TyMi, 2024-06-30:
+1
Same here on a WSL(V3). Receiving all messages via TCP but (nearly) no response is sent out. 
I had a bit more replies disabling meshtastic MQTT on WSL(V3) node. But far away from 100%...
Beside the already in other issues communicated errors, no specific error is raised after receiving and not sending reply.

Phlat, 2024-06-30:
The WaveShare Hat on the Pi MeshtasticD is a portduino version and works
well. I think it's the way the config interface stuff is handled trying to
work with both serial and tcp devices. OR it may even be my linux distro,
API version and libraries on this Raspberry Pi. For now I've given up using
the TCP node and went with a USB serial Heltec, which seems to be holding
up. I realize this is open source and am grateful to have it. It will be
fun to see what others bring to the project. 73

chicagolandmesh, 2024-07-04:
I just wanted to add that TCP works fine for me (Tested using an Ubuntu VM and on Docker).  The only thing that breaks it is if you reboot your node, then it won't automatically reconnect.  Maybe if that could be fixed then TCP would work just a bit better.  Thanks

rj45jack, 2024-07-10:
I am having this problem, it might be related to [this "Issue"](https://github.com/meshtastic/python/issues/594).