TheCommsChannel/TC2-BBS-mesh #31: Periodic errors

brad28b, 2024-07-01:
Great project! Love it.

I have found it very useful, just reporting the following errors, which occur every few mins, but don't seem to affect anything that kills any functionality. It seems to happen whether I connect via TCP or Serial, same error apart from the Interface.

2024-07-01 16:50:09,608 - ERROR - Unexpected error in deferred execution <class 'pubsub.core.topicargspec.SenderUnknownMsgDataError'>
Traceback (most recent call last):
  File "/home/brad/lib/python3.11/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/brad/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/brad/lib/python3.11/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/brad/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 433, in publish
    self._getListenerSpec().check(msgData)
  File "/home/brad/lib/python3.11/site-packages/pubsub/core/topicargspec.py", line 229, in check
    raise SenderUnknownMsgDataError(self.topicNameTuple,
pubsub.core.topicargspec.SenderUnknownMsgDataError: Some optional args unknown in call to sendMessage('('meshtastic', 'receive')', packet,interface): interface

ianmcorvidae, 2024-07-02:
Just linking identical issues together: This `SenderUnknownMsgDataError` is the same as #4 and is fixed by what's in the comment at: https://github.com/TheCommsChannel/TC2-BBS-mesh/issues/24#issuecomment-2198330622

TheCommsChannel, 2024-07-02:
Testing this fix now. I get these from time to time when I'm back on the main mesh.

TheCommsChannel, 2024-07-02:
Fix seems to be working and just updated the code with the fix. Thanks!

https://github.com/TheCommsChannel/TC2-BBS-mesh/commit/be9aa375cf0ed8f96fb2a51e37f87698ff26d36f

closed 2024-07-02 by TheCommsChannel

brad28b, 2024-07-03:
Confirming it's fixed for me, after updating