TheCommsChannel/TC2-BBS-mesh #33: HELP command not giving response

gretel, 2024-07-01:
[image]

think something is wrong with the state keeping but i haven't read the code in depth. regards

d42kph453, 2024-07-05:
Recommend closing this issue as the new menu updates have been applied to the git repository. This was identified as an iOS device issue as Androids did not experience this problem. Follow on issue #37 also identified this issue.

Recommend closing this issue as the new menu updates have been applied to the git repository. Users have confirmed that the menus are displaying the initial main menu.

gretel, 2024-07-06:
> Recommend closing this issue as the new menu updates have been applied to the git repository. 

sure, go ahead 😏

OHTechDiver, 2024-07-07:
Is "HELP" simply the main menu?  Since the server responds with the main menu to any input that's not a recognized command, is the menu item even necessary?  Similarly, does "EXIT" serve a function?