TheCommsChannel/TC2-BBS-mesh #34: [Feature Request]:  Translations

memo-567, 2024-07-02:
Great job :+1:  

Is there a way to translate strings and menus into other languages?

TheCommsChannel, 2024-07-02:
Danke 🙂

Not sure, but I can certainly look into it. I saw some other comments in https://github.com/TheCommsChannel/TC2-BBS-mesh/issues/5 saying they can help translate.

Ztrby, 2024-07-13:
i think a first step could be to make the code so if can handle all textstrings from a source, then we can just see what translations that the community will do. 
What we see from other translations is that only some of them are 100% so it should always revert back to English when a string are missing in the translation. 
It could also be a way to customize the text even in English.

memo-567, 2024-07-13:
It's not only the text strings, but also the menu commands (letters)

Some still fit, others need adjustment.

**(M)ail,** for example, can remain **(M)ail** in the German translation, as it is often called that here.

But then **(C)hannels** becomes **(K)anäle**  ¯\\\_(ツ)_/¯

Ztrby, 2024-07-13:
that complicates matters :/ . if its important that the command are part of the word, it could be [C] Kanäle
i'm swedish myself so the swedish translation will have the same issue

memo-567, 2024-07-13:
Of course, it would be possible to use numbers instead of letters. All in all, there is no association to make it easier to remember.

It might make more sense to use placeholders in `message_proceessing.py` also.

This means that almost any language and any shortcut could be used.

Since I'm not a programmer, that's just a thought ¯\\\_(ツ)_/¯

Ztrby, 2024-07-13:
I guess it's easier to prepare for it now when its just a couple of hundreds of line of code then do it later when its just so mush more code. but it's prioritizations.  i'm not a python programmer either. but looking in to the code it looks like something i could learn and contribute something too.

qbak, 2024-11-26:
Hi, I would like to help translating into Polish... Is the idea alive? Thanks. Best