TheCommsChannel/TC2-BBS-mesh #36: [Feature Request] Waypoint Map Bulletin

infered5, 2024-07-03:
Add a way to mark waypoints in a bulletin with user-defined lists. 

Waypoints are pinned locations on the Map, which can be arbitrarily set to any GPS coordinate. They have three major features: GPS Coordinates, Name, and Description.

Under ideal conditions, the user could open a bulletin for waypoints, select their category, and either get every Waypoint sent, or individual ones in a list. Waypoints should be categorized arbitrarily and should be appended if necessary. 

Example Waypoint categories: Fishing Spots, LBGT friendly businesses, Repeater Locations, etc.

**Challenges:**
* As the Android app currently stands, waypoints are only sent via default channel. Someone with an iOS device will have to confirm this for that platform. For most users, this is the unencrypted LongFast. Pins _may_ have to be worked on via the firmware/app side in order to DM them to the bulletin node.

* If several waypoints are sent at once, this may flood the network. Each waypoint that gets sent is its own message, and this may necessitate the exclusion of a "send all waypoints from this category" option.

* An edited waypoint will not override on existing maps, and if a typo is fixed or the title/description edited later for any reason, there will be two waypoints on the client app on the same location.

infered5, 2024-07-03:
The devs have confirmed with me that (unknown if intentional), waypoints are always sent via primary channel on both iOS and Android apps. They cannot be sent via DM. This means as it is, waypoints when sent by anybody, or if anybody retrieves a waypoint from the BBS, it will show up for everyone, assuming the BBS is on the primary channel. This may be a dealbreaker for this FR unless we can have the app modified.

TheCommsChannel, 2024-07-03:
This won't work for the mesh version of the BBS since it's mostly interacted with via DM. Something similar is planned for the APRS version though.

closed 2024-07-03 by TheCommsChannel