TheCommsChannel/TC2-BBS-mesh #37: No "Menu" opinion is received?

lostngone, 2024-07-04:
I can type "Hello" or "H" and I get no response from the BBS, however other commands do return a response.
[image]

I up'ed the logging on the server.py to debug(see attached). My first send was "Hello", then "H" and I didn't see any responses to either, then I sent "F" and did see the fortune. Inside the debug I do not see any clear errors on "Hello" or "H" however I am no where near the level to understand any of it.
[serverdebug.txt](https://github.com/user-attachments/files/16091755/serverdebug.txt)

I am running this on a Raspberry pi 4 with a clean install of Raspberry Pi Bookworm 64-bit and followed the directions to from the link below to install TC2-BBS.

https://github.com/TheCommsChannel/TC2-BBS-mesh

I am connecting to the BBS node via serial and it is a T-Lora V2 running 2.3.13.

TheCommsChannel, 2024-07-04:
Based on reports, this is looking to be on the iOS side. No reports of issues on Android. I wonder if it's the █▓▒░ TC² BBS ░▒▓█  banner or something it doesn't like.

Could you try removing it from the banner. Its line 40 in `command_handlers.py` Maybe just change it to BBS or something simple to see if that fixes it

SirMcQuery, 2024-07-04:
I had the same issue, the weird thing is i got it to send the menu message once. After changing the "█▓▒░ TC² BBS ░▒▓█" I am getting consistent help menu replies with both H and Help.

lostngone, 2024-07-04:
I changed line 44(title = ) to just plan text and it has improved things greatly. I am still seeing times when the server seems to not return the menu, restarting the server seems to fix the issue.

THANK YOU for everyone that worked this issue, I owe you a beer!.

d42kph453, 2024-07-05:
Recommend closing this issue as the new menu updates have been applied to the git repository.

TheCommsChannel, 2024-07-09:
This should be fixed now with the new menu updates

closed 2024-07-09 by TheCommsChannel