TheCommsChannel/TC2-BBS-mesh #39: BBS Message Syncing Actions (Delete Action)

d42kph453, 2024-07-04:
I am new to posting GitHub errors/feedback. Please feel free to direct me to enter data as you desire to better assist in troubleshooting.

An error occurred when syncing a delete action across BBS nodes. The message being existed on node 1 or 3 prior to nodes 2 and 3 being added.  Configuration set is:
Node 1: TBeam running firmware Alpha 2.3.14.xx, TC-BBS (2024-7-3 pull + manual BBS title change)
Node 2: Heltec v3 running firmware Beta 2.3.13.xx, TC-BBS (2024-7-3 pull + manual BBS title change)
Node 2: Heltec v3 running firmware Beta 2.3.13.xx, TC-BBS (2024-7-3 pull + manual BBS title change)

Note: Manual title change is allow menu to be visible to iOS devices. Still testing, but special characters and graphics cause issues. Using Alpha-Numeric only seems to work title is shorter.

The syncing feedback from node 1:
2024-07-03 23:14:40,380 - INFO - Attempting to delete mail with unique_id: e3062fc4-a781-44ea-b640-9eabdd251b27 by !335c1830
2024-07-03 23:14:40,395 - INFO - SERVER SYNC: Sending delete mail sync message with unique_id: e3062fc4-a781-44ea-b640-9eabdd251b27
2024-07-03 23:14:50,927 - INFO - Mail with unique_id: e3062fc4-a781-44ea-b640-9eabdd251b27 deleted and sync message sent.

node 2:
2024-07-03 23:14:46,092 - INFO - Received message from user 'QBBS' to KI61: DELETE_MAIL|e3062fc4-a781-44ea-b640-9eabdd251b27
2024-07-03 23:14:46,092 - INFO - Processing delete mail with unique_id: e3062fc4-a781-44ea-b640-9eabdd251b27
2024-07-03 23:14:46,093 - ERROR - Unexpected error in deferred execution <class 'TypeError'>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/usr/local/lib/python3.9/dist-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/opt/TC2-BBS-mesh/server.py", line 61, in receive_packet
    on_receive(packet, interface)
  File "/opt/TC2-BBS-mesh/message_processing.py", line 89, in on_receive
    process_message(sender_id, message_string, interface, is_sync_message=True)
  File "/opt/TC2-BBS-mesh/message_processing.py", line 50, in process_message
    delete_mail(unique_id, [], interface)
TypeError: delete_mail() missing 1 required positional argument: 'interface'

node 3:
2024-07-03 23:15:01,330 - INFO - Received message from user 'QBBS' to KI62: DELETE_MAIL|e3062fc4-a781-44ea-b640-9eabdd251b27
2024-07-03 23:15:01,330 - INFO - Processing delete mail with unique_id: e3062fc4-a781-44ea-b640-9eabdd251b27
2024-07-03 23:15:01,330 - ERROR - Unexpected error in deferred execution <class 'TypeError'>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/usr/local/lib/python3.9/dist-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/opt/TC2-BBS-mesh/server.py", line 61, in receive_packet
    on_receive(packet, interface)
  File "/opt/TC2-BBS-mesh/message_processing.py", line 89, in on_receive
    process_message(sender_id, message_string, interface, is_sync_message=True)
  File "/opt/TC2-BBS-mesh/message_processing.py", line 50, in process_message
    delete_mail(unique_id, [], interface)
TypeError: delete_mail() missing 1 required positional argument: 'interface'

d42kph453, 2024-07-04:
Similar error occurred during syncing from node 2 <> 1 and 3.  

This was the message:
2024-07-03 23:47:53,550 - INFO - Received message from user 'QDB2' to KI61: This is a test message from your friendly bbs
2024-07-03 23:48:07,622 - INFO - Received message from user 'QDB2' to KI61: Here is a 100-character sample text: "Concise writing conveys ideas clearly, ensuring readers grasp concepts quickly and effectively."
2024-07-03 23:48:25,161 - INFO - Received message from user 'QDB2' to KI61: End

Error on node 2 and 3 during sync attempt:
2024-07-03 23:48:41,859 - INFO - Received message from user 'KI61' to KI62: MAIL|!335d8ebc|QDB2|!da5ebe60|This is a test message from your friendly bbs|Here is a 100-character sample text: "Concise writing conveys ideas clearly, ensuring readers grasp concepts quickly and eff
2024-07-03 23:48:41,859 - ERROR - Unexpected error in deferred execution <class 'IndexError'>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/usr/local/lib/python3.9/dist-packages/meshtastic/mesh_interface.py", line 1176, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/usr/local/lib/python3.9/dist-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/opt/TC2-BBS-mesh/server.py", line 61, in receive_packet
    on_receive(packet, interface)
  File "/opt/TC2-BBS-mesh/message_processing.py", line 89, in on_receive
    process_message(sender_id, message_string, interface, is_sync_message=True)
  File "/opt/TC2-BBS-mesh/message_processing.py", line 42, in process_message
    sender_id, sender_short_name, recipient_id, subject, content, unique_id = parts[1], parts[2], parts[3], parts[4], parts[5], parts[6]
IndexError: list index out of range

2024-07-03 23:48:42,620 - INFO - Received message from user 'KI61' to KI62: ectively."
|c7afc984-75ef-4861-a214-bfc76a4cd005

TheCommsChannel, 2024-07-04:
This should be fixed now if you want to give it a go

d42kph453, 2024-07-04:
Syncing is working without displaying errors BUT....

I am seeing a possible side effect to the change. When I first used node 1 to send a message, syncing would occur on both node 2 and node 3. This is now not occurring. Syncing only occurs to node 1 or 2 or 3. I only see a sync occurring one time. Initially I thought this was because all my BBS nodes were set to CLIENT_MUTE, so I changed them all to CLIENT.

Same result as CLIENT_MUTE. The BBS will sync 1 time and 1 time only, even when 2 nodes are within range/proximity (1 foot). Regardless of the BBS node I use to send a message, only 1 of the remaining 2 nodes sees a SERVER SYNC action.

d42kph453, 2024-07-04:
Update:

I preformed additional tests. I cleaned up my sync_node list in the config.ini file to exclude the node id of the config.ini was running on.  I set up the some sync list across all my bbs nodes. It is possible that when the node tries to SERVER SYNC across the nodes sync list and sees itself, it ends the sync action.

Maybe there is a way to code the bbs_nodes list to automatically exclude itself if someone (like me) uses a list across nodes. Not sure this is important, but good to now that if synchronizations to other node is not occurring when they are in range, maybe they have the node sending the sync self-listed in the bbs_nodes listing.

d42kph453, 2024-07-05:
The error related to server sync actions has been resolved.

closed 2024-07-05 by d42kph453