TheCommsChannel/TC2-BBS-mesh #40: [Feature Request] Message Short Name - Upper/Lower Case Insensitivity

d42kph453, 2024-07-04:
I noted that when sending a new message, the short name is case sensitive. If entered in lowercase, a node is not found in database if the short name was in uppercase. For simplicity, allowing the short name would allow for less key strokes for entering data via mobile device.

Another suggestion for potential short name is to plan for duplicates and how to show multiple results. Perhaps this could be a numbered option menu if multiple are found with long name displayed for correct choice option. Maybe limit this to displaying 10 matches maximum, or less.

TheCommsChannel, 2024-07-04:
Good catch. I'll fix it to be case insensitive.

Handling of duplicates is already in place. Let me know if you're having issues with it though
`send_message("There are multiple nodes with that short name. Which one would you like to leave a message for?", sender_id, interface)`

d42kph453, 2024-07-04:
I have 3 BBS nodes that I can manually test solutions. Happy to support a great project. Let me know.

TheCommsChannel, 2024-07-04:
This has been implemented. Should be good but feel free to test on your end. Appreciate your support and suggestions!

d42kph453, 2024-07-04:
Confirmed. Prompt is now case insensitive. Thank you.

d42kph453, 2024-07-05:
This could be closed or expanded. The new menus introduce via a pull on 7/4/24 are case sensitive. Let me know if you prefer for me to open a new ticket, as this initial issue was resolved.

closed 2024-07-05 by d42kph453