TheCommsChannel/TC2-BBS-mesh #42: TCP connection timeout after node disconnects or reboots

chicagolandmesh, 2024-07-05:
TCP works great unless the node is disconnected from the internet or it reboots.  This causes the script to stop working.  This doesn't seem too hard to remedy.  Thanks

d42kph453, 2024-07-05:
@chicagolandmesh - Do you have any logged errors for the disconnect that you could post here?

vloschiavo, 2024-07-05:
Here's a sample log output

How to reproduce:     

-Docker image from the latest git commit [aa6f708](https://github.com/TheCommsChannel/TC2-BBS-mesh/commit/aa6f70831ff04e92f108a0dd2f74ab84e8163403) connected to the primary node over serial USB

-Primary Node (RAKWireless 4631)

-Meshtastic Android app connected to Second Node over TCP (Heltec Wireless Stick Lite v3)                                                                                     
-Monitor log of BBS: ```docker logs bbs -f```

-Test functionality with Fortune. Get response:
```2024-07-05 20:04:41,669 - INFO - Received message from user 'VLoR' to VLo1: F```
App shows fortune

-Issue a reboot over the admin channel to the BBS node.
Docker log output:
```2024-07-05 20:29:45,033 - WARNING - Meshtastic serial port disconnected, disconnecting... device reports readiness to read but returned no data (device disconnected or multiple access on port?)```

The node successfully reboots and can accept commands from the usb serial or from the admin channel. 
No further log output from the BBS
The BBS doesn't reconnect.

ddthekidd, 2024-07-06:
I'll gather the error message and comment it here when I do it.  I know that restarting the container will fix it but I think it could be easily fixed to avoid that.

chicagolandmesh, 2024-07-07:
Ok, I just tested it again and was able to get this error in my log

2024-07-06 22:37:02.679276-05:002024-07-07 03:37:02,678 - INFO - TC²-BBS is running on tcp interface...
2024-07-06 22:37:02.689741-05:002024-07-07 03:37:02,689 - INFO - Received message from user 'kidd' to DFIX: QCH
2024-07-06 22:42:02.696785-05:002024-07-07 03:42:02,695 - ERROR - Unexpected OSError, terminating meshtastic reader... [Errno 104] Connection reset by peer

chicagolandmesh, 2024-07-07:
It seems like it could be easy to implement a check to see if the node ip is alive or not and attempt to reconnect as opposed to shutting down the container.

chicagolandmesh, 2024-07-08:
I have not rebooted my node today and I noticed this in the logs 
[tc2-bbs-mesh_tc2-bbs-mesh-ix-chart-75fb55989f-slq78_ix-chart.log](https://github.com/user-attachments/files/16133684/tc2-bbs-mesh_tc2-bbs-mesh-ix-chart-75fb55989f-slq78_ix-chart.log)