TheCommsChannel/TC2-BBS-mesh #45: BrokenPipeError: [Errno 32] Broken pipe

out0fsync, 2024-07-06:
After fresh install I get this error:

2024-07-06 14:32:46,457 - INFO - TC²-BBS is running on tcp interface...
Database schema initialized.
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 1378, in run
    self.function(*self.args, **self.kwargs)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 726, in callback
    self._sendToRadio(p)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 786, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 1378, in run
    self.function(*self.args, **self.kwargs)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_int                                                                             erface.py", line 726, in callback
    self._sendToRadio(p)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 786, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
Exception in thread Thread-7:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 1378, in run
    self.function(*self.args, **self.kwargs)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 726, in callback
    self._sendToRadio(p)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 786, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
Exception in thread Thread-8:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 1378, in run
    self.function(*self.args, **self.kwargs)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 726, in callback
    self._sendToRadio(p)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 786, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe
Exception in thread Thread-9:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 1378, in run
    self.function(*self.args, **self.kwargs)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 726, in callback
    self._sendToRadio(p)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 786, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/root/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)
BrokenPipeError: [Errno 32] Broken pipe

out0fsync, 2024-07-08:
I pasted this code into the top of the 3 files listed in the log above.

from signal import signal, SIGPIPE, SIG_DFL  
signal(SIGPIPE,SIG_DFL) 

Does this suppress the error messages and allow the code to run without errors?

out0fsync, 2024-07-11:
when i run the BBS using serial rather tcp everything wroks without errors

Ztrby, 2024-07-14:
i think maybe issue 52 could be the same. 
BrokenPipeError: [Errno 32] Broken pipe is in the logs from the docker. 

https://github.com/TheCommsChannel/TC2-BBS-mesh/issues/52

StevoKeano, 2024-08-11:
For now, crontab job :  status checks every 60 seconds, restart service if I see BrokenPipeError: 

 mesh-bbs.service - TC²-BBS Meshtastic Version
     Loaded: loaded (/etc/systemd/system/mesh-bbs.service; enabled; preset: ena>
     Active: active (running) since Sat 2024-08-10 20:06:01 CDT; 10min ago
   Main PID: 24893 (python3)
      Tasks: 4 (limit: 762)
        CPU: 10min 14.553s
     CGroup: /system.slice/mesh-bbs.service
             └─24893 /home/stevo/TC2-BBS-mesh/venv/bin/python3 /home/stevo/TC2->

Aug 10 20:16:05 bbs1 python3[24893]:     self.function(*self.args, **self.kwarg>
Aug 10 20:16:05 bbs1 python3[24893]:   File "/home/stevo/TC2-BBS-mesh/venv/lib/>
Aug 10 20:16:05 bbs1 python3[24893]:     self._sendToRadio(p)
Aug 10 20:16:05 bbs1 python3[24893]:   File "/home/stevo/TC2-BBS-mesh/venv/lib/>
Aug 10 20:16:05 bbs1 python3[24893]:     self._sendToRadioImpl(toRadio)
Aug 10 20:16:05 bbs1 python3[24893]:   File "/home/stevo/TC2-BBS-mesh/venv/lib/>
Aug 10 20:16:05 bbs1 python3[24893]:     self._writeBytes(header + b)
Aug 10 20:16:05 bbs1 python3[24893]:   File "/home/stevo/TC2-BBS-mesh/venv/lib/>
Aug 10 20:16:05 bbs1 python3[24893]:     self.socket.send(b)
Aug 10 20:16:05 bbs1 python3[24893]: BrokenPipeError: [Errno 32] Broken pipe

stevo@bbs1:~ $ # Check the status of the mesh-bbs.service
sudo systemctl status mesh-bbs.service
● mesh-bbs.service - TC²-BBS Meshtastic Version
     Loaded: loaded (/etc/systemd/system/mesh-bbs.service; enabled; preset: ena>
     Active: active (running) since Sat 2024-08-10 20:17:01 CDT; 21s ago
   Main PID: 25135 (python3)
      Tasks: 4 (limit: 762)
        CPU: 17.328s
     CGroup: /system.slice/mesh-bbs.service
             └─25135 /home/stevo/TC2-BBS-mesh/venv/bin/python3 /home/stevo/TC2->

Aug 10 20:17:01 bbs1 systemd[1]: Started mesh-bbs.service - TC²-BBS Meshtastic >
Aug 10 20:17:04 bbs1 python3[25135]: 2024-08-10 20:17:04 - INFO - TC²-BBS is ru>
Aug 10 20:17:04 bbs1 python3[25135]: 2024-08-10 20:17:04 - JS8Call - INFO - JS8>
lines 1-12/12 (END)

enforcer2000, 2024-12-11:
_> For now, crontab job : status checks every 60 seconds, restart service if I see BrokenPipeError: see: https://github.com/StevoKeano/mestastic-utils/blob/main/check_and_restart_service.py
>_ 

Just set up this script and crontab job for my setup as I was having the same issues (running on a Pi Zero 2W with waveshare lora HAT - running as per the install instructions over TCP to localhost - many thanks for this solution as it has got me going - hopefully someone can get to the bottom of it for a permanent fix.. 

Thanks again!

StevoKeano, 2024-12-11:
Cool, glad it helps!
Steve 
Join AustinMESH.org
K3ANO Amateur License 
WSDI766  GMRS
-------- Original Message --------On 12/11/24 3:59 PM, enforcer2000  wrote:
_> For now, crontab job : status checks every 60 seconds, restart service if I see BrokenPipeError: see: https://github.com/StevoKeano/mestastic-utils/blob/main/check_and_restart_service.py

_

Just set up this script and crontab job for my setup as I was having the same issues (running on a Pi Zero 2W with waveshare lora HAT - running as per the install instructions over TCP to localhost - many thanks for this solution as it has got me going - hopefully someone can get to the bottom of it for a permanent fix..
Thanks again!

[
{
"target": "https://github.com/TheCommsChannel/TC2-BBS-mesh/issues/45#issuecomment-2537281973",
"url": "https://github.com/TheCommsChannel/TC2-BBS-mesh/issues/45#issuecomment-2537281973",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]