TheCommsChannel/TC2-BBS-mesh #47: Mail and Bulletin Menus Broken after Addition of Quick Commands

OHTechDiver, 2024-07-07:
The original Mail and Bulletin menus, using M and B respectively, do not seem to work after the recent addition of the Quick Commands (QCH).  Everything seems to work using the SM| format with short-name, subject,.  I have seriously mixed emotions about the QCH commands and use of pipes on iOS--requires three presses to navigate to the "|" on the default iOS keyboard.   [Windows 11,   Python 3.11.8, RAK4631, Meshtastic 2.3.14]

TheCommsChannel, 2024-07-07:
I'll be redoing the entire menu. As far as pipes, what is a better option for iPhone users? I'm on Android so I'm not familiar with the keyboard.

OHTechDiver, 2024-07-07:
> As far as pipes, what is a better option for iPhone users?

I really don't have a better suggestion.  To be fair, I probably would have used pipes too. Keep up the great work...this takes Meshtastic to another level.

TheCommsChannel, 2024-07-09:
Meu has been updated. Also, using two commas as a delimiter now instead of pipe. Comma is on the main Android keyboard. Not sure on iPhone.

closed 2024-07-09 by TheCommsChannel

1n5aN1aC, 2024-07-09:
> Meu has been updated. Also, using two commas as a delimiter now instead of pipe. Comma is on the main Android keyboard. Not sure on iPhone.

Comma isn't on the first main keyboard page, but it's easily accessible, unlike Pipe.