TheCommsChannel/TC2-BBS-mesh #50: [BUG] Message Recieve on different Channel

fti7, 2024-07-10:
I have two Channels configured on my personal + BBS Meshtastic Device.
Since i cant choose the "Channel" used for the DM on the Android App, it happens that i get the reply on the different Channel. And therefore it create a new Message Window.

https://github.com/meshtastic/Meshtastic-Android/issues/1125

Looks like you should also set the channelIndex on sendText, so the reply is on the same Channel as recieved
https://github.com/meshtastic/python/blob/3c772b5a31bee2170da8816fc73b0f14e99fcde1/meshtastic/mesh_interface.py#L319

rj45jack, 2024-07-10:
No channel is used since the BBS responds to DMs. The BBS should probably be running through it's own Meshtastic device.

fti7, 2024-07-10:
> No channel is used since the BBS responds to DMs. The BBS should probably be running through it's own Meshtastic device.

It is running on a own Meshtastic Device. Channels are still used for DMs by using its Encryption Key(s), even when sending to single Nodes. Check out the Meshtastic Android Issue

rj45jack, 2024-07-10:
> It is running on a own Meshtastic Device. Channels are still used for DMs by using its Encryption Key(s), even when sending to single Nodes. Check out the Meshtastic Android Issue

Oh interesting. For some reason in my head it wasn't.