TheCommsChannel/TC2-BBS-mesh #52: After a view hours no response anymore

wwbusch, 2024-07-12:
Hi Guys,

After a view hours the bbs is not responding anymore to commands. the meshtastic device is receiving data but seems that communication to the bbs stoped. its running on raspberry 4 with usb cable. any idea how to debug?
Wolf.

Ztrby, 2024-07-12:
I have the same problem, i connect over TCP/IP to my device.

TheCommsChannel, 2024-07-12:
> Hi Guys,
> 
> After a view hours the bbs is not responding anymore to commands. the meshtastic device is receiving data but seems that communication to the bbs stoped. its running on raspberry 4 with usb cable. any idea how to debug? Wolf.

What device role is your node in?

Ztrby, 2024-07-12:
my node is client

Ztrby, 2024-07-12:
[image]

wwbusch, 2024-07-12:
my node is router_client
i configured it to autostart the bbs to be reboot-ready. so i cant see any debug output on the screen...
Wolf.

Spogster, 2024-07-12:
I have exactly same problem.
Client role
Docker V1.0.29
Connecting over TCP/IP
The node is running on a Raspberry Pi with Meshtastic [2.3.15.deb7c27 Alpha] but tried it with [2.3.13.83f5ba0 Beta] with the same results.

Ztrby, 2024-07-12:
I start thinking maybe the error is on meshtastic inside the docker container because it logs error for files in "/usr/lib/python3.12/site-packages/meshtastic/"

TheCommsChannel, 2024-07-12:
> my node is router_client i configured it to autostart the bbs to be reboot-ready. so i cant see any debug output on the screen... Wolf.

router_client is the issue

router_client should rarely be used (and this role is actually going away https://github.com/meshtastic/firmware/issues/4185)

To get this working with router_client, changing the nodes sleep settings with `meshtastic --set power.ls_secs 600` should fix it. Be sure you're using router_client appropriately though. This should only be used for infrastructure nodes on top of mountains or tall towers, ect.

wwbusch, 2024-07-12:
its placed on a tv tower on my town so would cover 2 mio town. but i will change it to client mode and test again (maybe i will put 2nd device as a router and not connect it to the raspi and than put the BBS on not that good location. Update will follow

Ztrby, 2024-07-12:
I have client roll , tomorrow I will try to downgrade my docker build to a older one and see if I still have the problem.

Ztrby, 2024-07-13:
> I have exactly same problem. Client role Docker V1.0.29 Connecting over TCP/IP The node is running on a Raspberry Pi with Meshtastic [2.3.15.deb7c27 Alpha] but tried it with [2.3.13.83f5ba0 Beta] with the same results.

I will try Docker 1.0.25 today and if it still have problems i plan to go to 1.0.20, i post back about my results

wwbusch, 2024-07-13:
Ztrby: not really knowing much about linux: i followed the installation here on github which worked super easy for me. i did not notice that there was docker installed or its using a docker? is this standard and just not so easy to notice? as i understood the bbs is just using python?
Wolf. (i will now install a router device on the high location and the bbs with a client in the buttom of the tower...)

Ztrby, 2024-07-13:
> Ztrby: not really knowing much about linux: i followed the installation here on github which worked super easy for me. i did not notice that there was docker installed or its using a docker? is this standard and just not so easy to notice? as i understood the bbs is just using python? Wolf. (i will now install a router device on the high location and the bbs with a client in the bottom of the tower...)

Docker is a container that have everything inside it, so its like a image that you just start, so in theory everybody should have the same issue. so the 1.0.25 was running for 3 hours alredy. so lets se if it works better.

Ztrby, 2024-07-13:
So i had to restart because of an other problem, and i tested 1.0.20, it fast only up 13 min before crashing. 
So now i fleshed the nod with lates alpha 2.3.15 och docker at 1.0.10, so now i might run in to problem with the old menu system instead. only time can tell. if this works i redo all the previous testes with the new node firmware

Ztrby, 2024-07-14:
none of the above containers work, so now im back at the latest 1.0.29 and i restart it every hour. 
I maybe try another node like a heltec v3, if that fails maybe i try serial interface. 
if anybody solves is please share

Ztrby, 2024-07-14:
Can this be the same as issue 45? 
https://github.com/TheCommsChannel/TC2-BBS-mesh/issues/45

TheCommsChannel, 2024-07-14:
> Ztrby: not really knowing much about linux: i followed the installation here on github which worked super easy for me. i did not notice that there was docker installed or its using a docker? is this [...]

The BBS is meant to run on just python as I want to keep overhead low. Docker is optional for those that want to use it.

Ztrby, 2024-07-14:
so i don't think it's in the docker implementation anymore. thats why i use the latest again. issue 45 have the same problem in local install, so maybe we just found a bug in the python meshtastic API on tcp interface. then we have to move the issue to that git-project.  and a workaround could be to have some try catch and if the API stop responding we can restart the BBS until it fixed

TheCommsChannel, 2024-07-14:
> so i don't think it's in the docker implementation anymore. thats why i use the latest again. issue 45 have the same problem in local install, so maybe we just found a bug in the python meshtastic [...]

It could just be the BBS' implementation of it. I think there are a number of projects that use TCP with no issue, but I've not tested them to say for sure. 

You may want to try this one and see if this works via TCP for you
https://github.com/pdxlocations/meshtastic-curses-client

fti7, 2024-07-21:
I have the same issue. Moved from TCP to Serial Interface. Same issue
Strange is that i see the Message recieved from my Smartphone in the Docker Log. But i didnt get any reply.
Replies should be logged too. Maybe some indicator when recieved by the Meshtastic Device

pitbullcoder, 2024-07-22:
This can be power related too. Not all USB cables are equal by far. So try different cables and different power supplies. I had to pull out my box of cables and try different combinations initially before I had a stable power supply to both the RPI and the radio. Typically the thicker cables are better, and I try to keep them to 2-3ft max length

Ztrby, 2024-07-22:
I'm curious, how many with this issue have alpine Linux?

pitbullcoder, 2024-07-24:
> I'm curious, how many with this issue have alpine Linux?

I've been using RPI bookworm lite, no desktop just command line and it's been stable once I got my power issues and radio settings straight. Mine has been up for 2 weeks without shutdown/restart at the moment.