TheCommsChannel/TC2-BBS-mesh #56: Observed Main Menu Behavior

d42kph453, 2024-07-13:
Bluetooth Client being used: iOS

This may or may not be an issue, depending on the intended design.

In past, I would be able to send almost anything to have the BBS Menu prompt would be sent to me. If I sent "H", "Hello", or "Help" - the main menu option would appear. After the menu changes back to the single letter commands, I no longer get a menu response from the BBS node from the examples above.

I do receive the main menu options if I send "X" or "x" to a BBS node. This always works, regardless of location in the menu system.

If I send any on the main menu letters, they also respond. This is dependent on my session location. If a session is in [U]tils, then I would get no response until the command corresponds to the current menu section. I believe this is expected behavior in the sub menus. "X" always gets a return response.

"X" always works and receives a response, no matter where I am located in the menu system.

I just was not sure this was by design to not respond to "Help", or this was an unexpected side effect. Also, this is unknown to be an iOS only or multi-platform behavior.  I only have iOS at this time to test with.

d42kph453, 2024-07-14:
A second system setup the same way does return a menu. I will test a little more.  A new clean system prompts a menu when it receives external data directed to it.

pitbullcoder, 2024-07-15:
I see the same thing on my end, android client. X or x will return but H or help or anything else doesn't. Seems the fall through default response isn't working properly anymore. If I have time this week I'll see if I can get a fix with a PR soon.

pitbullcoder, 2024-07-15:
actually had a few minutes this morning, read through the code. I have a temporary patch for this, it doesn't resolve the root cause but it'll get the help command working and handle any garbage command returning the user back to main menu and present them with the list

PR incoming

pitbullcoder, 2024-07-15:
PR submitted: https://github.com/TheCommsChannel/TC2-BBS-mesh/pull/59

d42kph453, 2024-07-17:
I will take a look at it once applied. Thanks.

pitbullcoder, 2024-07-17:
Great, the feedback will be helpful

pitbullcoder, 2024-07-28:
> I will take a look at it once applied. Thanks.

If you haven't checked it yet, the issue back again after the JS8CALL was added to the menu. I've submitted another pull request to fix it again

https://github.com/TheCommsChannel/TC2-BBS-mesh/pull/71

Once this is merged we can update to latest and retest, but at the moment the bug is back. Really need to refactor the menu nav code to 'harden' it up. I've been too busy in work to do much other than submit a few patches to TC2