TheCommsChannel/TC2-BBS-mesh #58: Unexpected OSError ... [Errno 104] Connection reset by peer

ronnietucker, 2024-07-14:
Getting this error at random times:

ERROR - Unexpected OSError, terminating meshtastic reader... [Errno 104] Connection reset by peer
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 1394, in run
    self.function(*self.args, **self.kwargs)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 737, in callback
    self._sendToRadio(p)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 797, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)

BrokenPipeError: [Errno 32] Broken pipe
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 1394, in run
    self.function(*self.args, **self.kwargs)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 737, in callback
    self._sendToRadio(p)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 797, in _sendToRadio
    self._sendToRadioImpl(toRadio)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/stream_interface.py", line 115, in _sendToRadioImpl
    self._writeBytes(header + b)
  File "/home/ronnie/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/tcp_interface.py", line 73, in _writeBytes
    self.socket.send(b)

Ztrby, 2024-07-14:
i think this can be related to issue 45 och 52, do you use TCP? 

https://github.com/TheCommsChannel/TC2-BBS-mesh/issues/45
https://github.com/TheCommsChannel/TC2-BBS-mesh/issues/52

ronnietucker, 2024-07-14:
> i think this can be related to issue 45 och 52, do you use TCP?
> 
> #45 #52

Yes, using TCP.

I think it's due to the node losing internet for a moment.

Ztrby, 2024-07-14:
yes i do, i try some different things. i have 2 BBSs up now and one of them have been on för 5 hours now and it can be a new record for me, how long before you getting your error?

ronnietucker, 2024-07-14:
> yes i do, i try some different things. i have 2 BBSs up now and one of them have been on för 5 hours now and it can be a new record for me, how long before you getting your error?

Varies. Could me minutes, could be an hour.

I've got a Heltec running on serial just now. So we'll see how it goes there. That should be fine. I hope.  :D

Ztrby, 2024-07-14:
all people that had this problem run over tcp so far. 
I also had some error after just 10 min and sometimes run for hours. now i have 2 BBSs up and running so it will be easier to see if anyting i do helps

Daedalus-, 2024-07-28:
I can confirm this is happening via TCP and is triggered when the node restarts. (As with a config change) I haven't tested the node losing network connection scenario.

I would expect the application to crash entirely, so that an external watchdog or docker restart policy can recycle the app. Please feel free to reach out if I can help troubleshoot in any way.

RaspberryProgramming, 2025-11-06:
I wanted to mention the following issue on the meshtastic python repo https://github.com/meshtastic/python/issues/765