TheCommsChannel/TC2-BBS-mesh #66: Mesh to js8call

sfraise, 2024-07-19:
I understand the reasoning why this was setup as receive only from js8call to mesh and not mesh to js8call.  Potential legality issues and such.  Although I say that if a message is transmitted from my radio under my call it's legal and my responsibility.  I'd also say that if you have the skillset and radio to build this then you most likely have at least a general license.

However, I think there's a middle ground here.

In the js8call menu flow you could ask for a password that someone could set in the js8call part of the config so only trusted people on the mesh network could send.  You could also ask for callsign and have a list of trusted callsigns on the config side.  In fact, trusted callsigns could basically act as the password and simply just ask for callsign.

I think limitting js8call to receive only undercuts the power of what this could do.

I'd also add an option to select a non-longfast channel as the js8call encryption and encrypt the message on the pi using that encryption code which could then be decrypted on the receive side.  As far as I know data streams over hf are already encrypted by nature and legal so I don't think there's an issue with this.  This would further secure the message being sent over js8call where as it sits now anyone could read.

TheCommsChannel, 2024-07-19:
FCC won't allow retransmission from an unlicensed station unless it's weather or propagation info.

> (c) No station shall retransmit programs or signals emanating from any type of radio station other than an amateur station, except propagation and weather forecast information intended for use by the general public and originated from United States Government stations, and communications, including incidental music, originating on United States Government frequencies between a manned spacecraft and its associated Earth stations. Prior approval for manned spacecraft communications retransmissions must be obtained from the National Aeronautics and Space Administration. Such retransmissions must be for the exclusive use of amateur radio operators. Propagation, weather forecasts, and manned spacecraft communications retransmissions may not be conducted on a regular basis, but only occasionally, as an incident of normal amateur radio communications.

The same encryption rules apply on HF; there is no encrypted ham radio communication on HF

sfraise, 2024-07-19:
You're right, I thought about this last night laying in bed and realized I was wrong on that train of thought since it asctually rebroadcasting a radio signal.

Sooo, in config.ini let's add a list of allowed nodes and their callsigns if they have one.  Basically expanding on what you've already done with the urgent board concept.
Example:
[allow_list]
allowed_nodes = 12345678:KI1LEM, 87654321:KA1LME, 5555555555:0, ETC:ETC

[allow_options]
# bbs_allowed_option = allowed_nodes_only | all
bbs_allowed_option = allowed_nodes_only

Option to only allow those trusted nodes to interact with bbs or to allow anyone, then an option to allow transmit to js8call if a trusted node AND a callsign is added OR to leave it as receive only.
Example:
[js8call]
# enable_send = True | False (False by default)
enable_send = True

In the script you just break the allowed_node sting apart at the ":" or you could just make them json objects which would be easier to work with.

IF bbs_allowed_option == allowed_nodes_only AND {node_id} IN allowed_nodes:
    # continue on to bbs
ELSE:
    # break/return/log/whatever

Same concept as above for sending to js8call except this time you're checking if send enabled and checking against if the allowed_node has a callsign attached to it.  If yes show SEND menu option, if not then don't show.  You could even put a check against that again in the actual send function to be save just in case someone somehow was able to get to the send function.  No callsign return notice not allowed and break/return/whatever.

As for encryption on HF you're correct, "encoding" is where my head was at for some reason.  As long as you make the hashing/encoding public you should be golden as far as legality, however encrypting like the mesh does with a private key is not kosher.  Js8call by nature is already "encoded" (not encrypted like I implied in previous post).

So yes, no encrypting.