TheCommsChannel/TC2-BBS-mesh #75: Check board quick command

wishbone1138, 2024-08-04:
CB,,General or any other board does not appear to do anything for me.  It just tells me that no bulletins are available on the general board but when I go there directly I can clearly see one there.  Am I using this QC wrong?

kizeren, 2024-08-23:
This happens to not only pb,, but also for adding channels.  Everything gets set to lower case instead of case SEsnsItiVe.

kizeren, 2024-08-23:
This line `message = message.lower().strip()
` Breaks most input in the message due to it converting caps to lower.

Using `elif choice == 'p' or choice =='P':
` instead of `elif choice =='p':`  Should fix this.

Would have to do this for any input that uses message.lower.
I should have sometime this weekend to make the changes and do a bit of testing before I do a pull request.