TheCommsChannel/TC2-BBS-mesh #76: Unexpected error in deferred execution <class 'TypeError'>

mikerayborn, 2024-08-04:
I was testing your TC2-BBS software and it looks like there may be a small bug in the code for [N]odes:

2024-08-04 17:35:37 - INFO - Received message from user 'SLR2' to BBS: U
2024-08-04 17:35:37 - INFO - REPLY SEND ID=3211932453
2024-08-04 17:35:48 - INFO - Received message from user 'SLR2' to BBS: F
2024-08-04 17:35:48 - INFO - REPLY SEND ID=3211932454
2024-08-04 17:35:58 - INFO - Received message from user 'SLR2' to BBS: S
2024-08-04 17:35:58 - INFO - REPLY SEND ID=3211932455
2024-08-04 17:36:07 - INFO - Received message from user 'SLR2' to BBS: N
2024-08-04 17:36:07 - ERROR - Unexpected error in deferred execution <class 'TypeError'>
Traceback (most recent call last):
  File "/home/tc2/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/util.py", line 283, in _run
    o()
  File "/home/tc2/TC2-BBS-mesh/venv/lib/python3.10/site-packages/meshtastic/mesh_interface.py", line 1190, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
  File "/home/tc2/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/home/tc2/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/home/tc2/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/home/tc2/TC2-BBS-mesh/venv/lib/python3.10/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/home/tc2/TC2-BBS-mesh/server.py", line 71, in receive_packet
    on_receive(packet, interface)
  File "/home/tc2/TC2-BBS-mesh/message_processing.py", line 201, in on_receive
    process_message(sender_id, message_string, interface, is_sync_message=False)
  File "/home/tc2/TC2-BBS-mesh/message_processing.py", line 144, in process_message
    handle_stats_steps(sender_id, message, step, interface)
  File "/home/tc2/TC2-BBS-mesh/command_handlers.py", line 138, in handle_stats_steps
    total_nodes = sum(1 for node in interface.nodes.values() if node.get('lastHeard', 0) >= time_limit)
  File "/home/tc2/TC2-BBS-mesh/command_handlers.py", line 138, in <genexpr>
    total_nodes = sum(1 for node in interface.nodes.values() if node.get('lastHeard', 0) >= time_limit)
TypeError: '>=' not supported between instances of 'NoneType' and 'int'

2024-08-04 17:36:29 - INFO - Received message from user 'SLR2' to BBS: H
2024-08-04 17:36:29 - INFO - REPLY SEND ID=3211932456
2024-08-04 17:36:32 - INFO - REPLY SEND ID=3211932457
2024-08-04 17:36:51 - INFO - Received message from user 'SLR2' to BBS: R
2024-08-04 17:36:51 - INFO - REPLY SEND ID=3211932458
2024-08-04 17:36:53 - INFO - REPLY SEND ID=3211932459
2024-08-04 17:36:59 - INFO - Received message from user 'SLR2' to BBS: X
2024-08-04 17:36:59 - INFO - REPLY SEND ID=3211932460
--Mike Rayborn

Spogster, 2024-08-08:
I have been having the same issue it is always when non of the nodes have times against the seen list. If the nodes have times it works correctly. Using Raspberry PI with Waveshare Hat it always fails.

AndreasFeldt, 2024-10-10:
I have posted a pull request with a fix for this problem. 
In the meantime, you can edit the "command_handlers.py" file on line 138 to the following
`total_nodes = sum(1 for node in interface.nodes.values() if node.get('lastHeard', 0) is not None and node.get('lastHeard', 0) >= time_limit)
`
The `node.get('lastHeard', 0) is not None` part will remove all nodes with a NoneType or null value lastHeard field