TheCommsChannel/TC2-BBS-mesh #85: [question] Integration with Raspberry Pi as a standalone Meshtastic node

przemo, 2024-08-28:
Unless I completely misunderstand the idea , do you think it would be possible to integrate the BBS with standalone raspberry pi based Meshtastic node? Based on the Meshtastic info the Pi nodes are tested/worked with SPI only LoRa devices. 

Great work !!!

dd8aka, 2024-09-14:
Not sure what you exactly mean, but running Meshtasticd and TC2-BBS together on the same Raspberry Pi is no problem. Just use 

```
[interface]
type = tcp
hostname = 127.0.0.1
```

in your configuration of TC2-BBS

wabudd1, 2025-06-06:
Resurrecting this in case this is helpful:  I am running TC2 BBS on the same Pi as a Mesh node (using meshtasticd).  The TCP port used depends on what's in your `/etc/meshtasticd/config.yaml`.  I've customized my port to look like this:

```
Webserver:
  Port: 8043 # Port for Webserver & Webservices
```

This is what my `config.ini` looks like:

```
[interface]
type = tcp
hostname = 127.0.0.1
port = 8043
```