TheCommsChannel/TC2-BBS-mesh #86: [Question] Admin tool to post bulletins

hastur51, 2024-08-28:
Would be nice to have an admin tool to post bulletins from the server. I get we need to be aware they would have to be in character segments of 228 for Meshtastic. But this would be a helpful tool to post news bulletins.

on4rf, 2024-10-20:
I think this is a good idea. Doing so we cab add bulletins from a computer would save time and effort

typicalaimster, 2025-03-24:
Soo basically this. But highly modded right now for EDC Las Vegas and I'm still working out bugs. 

[image]

If there's interest it'd have to wait until after EDCLV for me to backport. Unless someone else wants to fish it out of my fork and merge the appropriate changes in. When it comes to the core BBS part I've tried to leave as much of that alone as possible.