TheCommsChannel/TC2-BBS-mesh #90: [Feature Request] Cross node connectivity

mertensm72, 2024-09-26:
I would like to use one BBS with two nodes: one 868 and one 433
My idea is to send BBS mails between different networks.
So the BBS has to remember the origin of each Node...

Bcpratt2000, 2024-09-30:
+1 to this, I would like to have the same BBS available on longfast/mediumfast/verylongslow

jabez007, 2025-01-02:
If you have multiple radios plugged in on different serial ports I think you could run multiple instances of the python server all connecting to the same sqllite db?

I don't have multiple radios, but if you do it should be simple to test if I'm understanding the code base:
In one terminal run `python server.py --interface-type serial --port $LongFastPort`
In another terminal run `python server.py --interface-type serial --port $MediumFastPort`

If it doesn't work, I would guess it to likely be a locking issue somewhere.

AdvJosh, 2025-03-04:
@jabez007 is correct; however, for this to be effective, there needs to be changes to the code to manage the SQLite db being locked out if another instance of the server is currently accessing it. One thing SQLite is not great for is concurrent connections. We would also need to add a config parameter for where the SQlite database is stored to allow multiple instances to connect to the same database without the end user modifying the code 

My simple way of dealing with concurrent connections would be to have the DB write commands be contained in a while true loop with a try/except  to handle if an error is thrown due to the database being locked. Basically, try to execute the insert/delete command and exit the loop if it is successful. If the write command isn't successful, sleep for a brief period and try again until it succeeds. 

Does anyone else have a better idea for implementation?