TheCommsChannel/TC2-BBS-mesh #95: Sometimes no Response of using Menu

Neppumuck, 2024-11-03:
Hi,
i installed the BBS on raspberry pi 4. I don´t use Docker. Cloned the GIT yesterday.
I can use the BBS, but i don´t get an answer when i do follow things:

Quick Commands:
CM = OK
SM = i get TC2 BBS Menu
PB = i get TC2 BBS Menu
CB = i get TC2 BBS Menu
Maybe my mistake - don´t know the correct syntax?

BBS Menu:
Mail = all R,S,X work fine
BBS = all G,I,N,U work fine (BTW how can Post be delete or changed?)
**Channel = V: OK, P: i dont´get an answer. In Debug see no Reaction. He see the "P" or "p" but nothing happens.**

Utilities Menu:
**Stats = H,R work fine, N: I get an error that i Post below**
Fortune = OK
Wall of Shame= OK

Please excuse me if I expressed myself in a complicated manner. Can you please help find the errors? Thank you in advance
Greetings Neppu

The Code when i use Stats from Nodes:
```
2024-11-03 14:27:21 - ERROR - Unexpected error in deferred execution <class 'TypeError'>
Traceback (most recent call last):
  File "/root/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/util.py", line 297, in _run
    o()
  File "/root/TC2-BBS-mesh/venv/lib/python3.11/site-packages/meshtastic/mesh_interface.py", line 1383, in <lambda>
    lambda: pub.sendMessage(topic, packet=asDict, interface=self)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/TC2-BBS-mesh/venv/lib/python3.11/site-packages/pubsub/core/publisher.py", line 216, in sendMessage
    topicObj.publish(**msgData)
  File "/root/TC2-BBS-mesh/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 452, in publish
    self.__sendMessage(msgData, topicObj, msgDataSubset)
  File "/root/TC2-BBS-mesh/venv/lib/python3.11/site-packages/pubsub/core/topicobj.py", line 482, in __sendMessage
    listener(data, self, allData)
  File "/root/TC2-BBS-mesh/venv/lib/python3.11/site-packages/pubsub/core/listener.py", line 237, in __call__
    cb(**kwargs)
  File "/root/TC2-BBS-mesh/server.py", line 71, in receive_packet
    on_receive(packet, interface)
  File "/root/TC2-BBS-mesh/message_processing.py", line 203, in on_receive
    process_message(sender_id, message_string, interface, is_sync_message=False)
  File "/root/TC2-BBS-mesh/message_processing.py", line 146, in process_message
    handle_stats_steps(sender_id, message, step, interface)
  File "/root/TC2-BBS-mesh/command_handlers.py", line 138, in handle_stats_steps
    total_nodes = sum(1 for node in interface.nodes.values() if node.get('lastHeard', 0) >= time_limit)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/TC2-BBS-mesh/command_handlers.py", line 138, in <genexpr>
    total_nodes = sum(1 for node in interface.nodes.values() if node.get('lastHeard', 0) >= time_limit)
                                                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: '>=' not supported between instances of 'NoneType' and 'int'
```

Neppumuck, 2024-11-04:
command_handlers.py / Line 383 must be :         elif choice.lower() == 'p':

enforcer2000, 2024-11-30:
I have a similar issue for the quick commands and also when I enter "Q" - I don't get the quick commands menu which I believe is supposed to show the syntax. It appears to be sent out to the nodes when I check the bbs log but never arrives at any of my meshtastic devices so I wondered if the message itself is corrupted somehow - any other thoughts? (The code correction above helped for that specific instance regarding the "P" request, and I don't have the "N" issue in the stats menu on my install - it comes back with the node information as expected)

This is a great project - thanks! I wish my Python knowledge was better as I would love to have a utility option "C" for chat - which then allows you to send a message to a local ollama/GPT instance at your base node which will send back the LLM response.

Neppumuck, 2024-12-01:
Hello enforcer2000,
I was able to observe that a few times too. If I use ESP32 devices, the error rarely occurs. I use nRF more often. In addition, the response time is significantly longer. Can you confirm that? Best regards

enforcer2000, 2024-12-01:
I am trying to use a really streamlined setup so I have meshtasticd running on the same pi zero2w as the TC2 BBS - using the waveshare lora hat with GPS. It is really neat and compact and obviously standalone. It works great for the other commands - sometimes it seems slow or other commands don't come through but I put this down to the mesh rather than the BBS. The N command works fairly consistently. What I can say though is that the Q command is never received at any of my other meshtastic devices - not even after a long period of time. I can see it sent in the Mesh-BBS logs though similar to all the other commands. I also only ever get the Main menu for the quick commands mentioned above (SM, PB, CB) as you have experienced.

TheCommsChannel, 2025-02-14:
Q command is working for me

Just sending SM PB and CB will not work. To see the example commands, send `SM,`, `PB,`, and `CB,,`

closed 2025-02-14 by TheCommsChannel

TheCommsChannel, 2025-02-14:
Sure thing!

That is odd about the Q command not working though... Works fine when I try it. I'll try to think if what could cause that. What LoRa hat are you using?

enforcer2000, 2025-02-17:
Hi @TheCommsChannel - So...

I have done some more testing and made some progress but also found some updates in the documentation on the Meshtastic website.

Firstly... I have now managed to get the Q command working. Generally.. it works if I spam the Q command at least 4 times as quickly as the Meshtastic app will let me. If I leave 10 seconds or more between each command then it doesn't seem to work,,, but... I have a feeling I think I have got to the issue and I believe this is down to the hardware setup I have.

I am using a Pi Zero with a Waveshare LoRaWan board - SX1262 LoRaWAN Node Module Expansion Board for Raspberry Pi (850~930MHz HF with GNSS)

Following a revisit to the Meshtastic site - it turned out this previously recommended board has now been picked up as having an issue - see the video here:
https://www.youtube.com/watch?v=PND1GlMSrEM

It's a shame as this was a really neat compact and low powered (or at least as low powered as I could think make a fully standalone BBS module) solution for an all-in-one TC2 BBS - but this to me seems the likely candidate for the cause of the issue - hopefully an updated hat in similar form factor will become available and readily accessible soon!

Anyway - I wanted to give a detailed response to help get to the bottom of the issue in case others have tried and had similar issues.

Thanks again for a fantastic project - I hope the concept continues to evolve with new features - I am still keen to put an offline LLM on the back of the BBS based on a specific menu trigger to provide useful responses to specific queries- I think the speed of a RPi 5 running a basic LLM is probably just about fast enough to provide short responses on the meshtastic protocol... given the protocol itself takes a few seconds to respond. To link the BBS menu to an Ollama engine would be great :)