TheCommsChannel/TC2-BBS-mesh #96: RPi zero USB

g7kse, 2024-11-09:
Not an issue with the software but a note that hopefully helps others. 

I had attempted to connect a Heltec V3 module (USB C) to a RPi zero W v1.1 and whilst it took a while to sort out there was an issue using the OTG cables that are commonly used to go from the micro USB to USB A. It wasn't happy. Not really solved it so will switch to an adapter cable that isn't OTG.

In the meantime it works nicely on an old RPI3...cracking bit of work

CcRup, 2024-11-29:
Running this BBS successfully on a R Pi Zero 2W. I've tried it with a RAK4631, T-echo and Heltec t114, individually. Verified serial function on all 3 devices through 2 different USB-C to Micro USB cables. The Pi Zero still can't see any radio (several different serial device not found errors when booting up the BBS). BBS setting is default at type = serial for one device at a time. Device settings verified with Client role and serial enabled. Went through the instructions 3 times with the published setup video. Don't think I missed anything. Any help would be greatly appreciated. Thanks!

jakern, 2024-11-30:
I can confirm I have this running on a zero w 1.1 with a rak4631. I'm not sure what I did differently.  Can you see the devices at /dev/tty*? Are you familiar with the common serial permissions issues? Just to check, only one of the usb micro ports uses OTG. Are you using the 'inner' one?

g7kse, 2024-12-03:
I'm not really sure what the issue is tbh. It's all very stable on the RPi 3 and there's no reason I can think of why it's not working as keyboard is fine for example. Lots of different cables tried as well but no luck. 

RPi is powered through the upper USB and the only thing I can think of is a lack of power to the Heltec. Just enough below a critical level.

CcRup, 2024-12-05:
> I'm not really sure what the issue is tbh. It's all very stable on the RPi 3 and there's no reason I can think of why it's not working as keyboard is fine for example. Lots of different cables tried [...]

It is possible to be a power delivery issue as the v3 uses an ESP32 chip. Can you try a Nordic based device like the ones I mentioned? I'm going to try a more robust power supply as well and report back. Unrelated, but I recall the pi 4 had an issue with lower quality power supplies.

CcRup, 2024-12-06:
> I can confirm I have this running on a zero w 1.1 with a rak4631. I'm not sure what I did differently. Can you see the devices at /dev/tty*? Are you familiar with the common serial permissions issues? Just to check, only one of the usb micro ports uses OTG. Are you using the 'inner' one?

Good to hear your set up is working!

- How can I look up that device address you're referencing?
-  I am not familiar with common serial issues and troubleshooting them further.
-  Also yes, I'm using the inner micro USB for the radio and the outer end one for power.

CcRup, 2024-12-06:
> Good to hear your set up is working!
>
> * How can I look up that device address you're referencing?
> * I am not familiar with common serial issues and troubleshooting them further.
> * Also yes, I'm [...]

Did some research on how to use Python and found the lsusb command. It shows my radio on bus 1 device 3. Not sure yet if this is relevant but the serial errors are gone. I only changed to a higher current power supply. Still no response from the BBS when messaging the node even though it sees it!
[image]

williamlambert, 2025-03-06:
> Running this BBS successfully on a R Pi Zero 2W. I've tried it with a RAK4631, T-echo and Heltec t114, individually. Verified serial function on all 3 devices through 2 different USB-C to Micro USB [...]

Could it be power related?