TheCommsChannel/TC2-BBS-mesh #99: TC2 BBS crashes after receiving JS8 heartbeat

FoxnoTango, 2024-11-29:
After starting the TC2 BBS software it connects to a Meshtatstic node on /dev/ttyUSB0 and to the JS8CALL software on the same Raspberry Pi where the TC2 BBS software is installed.
When receiving a heartbeat message on JS8 the TC2 BBS crashes:

_Configured to sync with the following BBS nodes: []
Nodes with Urgent board permissions: []
2024-11-29 16:48:59 - INFO - TC²-BBS is running on serial interface...
Database schema initialized.
2024-11-29 16:48:59 - JS8Call - INFO - Database tables created or verified.
2024-11-29 16:48:59 - JS8Call - INFO - Connecting to ('127.0.0.1', 2442)
2024-11-29 16:49:30 - JS8Call - INFO - Received JS8Call message: SM4MJR: to 3PC/4COA0O - HEARTBEAT SNR -19 ♢
Traceback (most recent call last):
  File "/home/beheer/TC2-BBS-mesh/server.py", line 93, in <module>
    main()
  File "/home/beheer/TC2-BBS-mesh/server.py", line 80, in main
    js8call_client.connect()
  File "/home/beheer/TC2-BBS-mesh/js8call_integration.py", line 206, in connect
    self.process(message)
  File "/home/beheer/TC2-BBS-mesh/js8call_integration.py", line 169, in process
    self.insert_message('messages', sender, receiver, msg)
  File "/home/beheer/TC2-BBS-mesh/js8call_integration.py", line 129, in insert_message
    ''', (sender, receiver_or_group, message))
                  ^^^^^^^^^^^^^^^^^
NameError: name 'receiver_or_group' is not defined_

Is this something is the software of the TC2 BBS or have I made a configuration error?

73 Foeke PA3FNT

myerika, 2024-12-19:
Is your problem solved?
73
Thomas SM0YOS

TheCommsChannel, 2024-12-21:
I'll have to do some testing. Seems like another user is having the same issue.

FoxnoTango, 2025-03-25:
This solved my problem:

From one of my Smarter than Me buddies:

For the TC2-BBS-mesh project:
in the js8call_integration.py file, change the following lines from/to:
line 129: Change:
''', (sender, receiver_or_group, message))
to
''', (sender, recipient, message))