TheCommsChannel/TC2-BBS-mesh #102: Stop after message from JS8CALL

myerika, 2024-12-14:
I got these message when receiving a message from JS8CALL and TC2-BBS-mesh stop, I have @HB in my group list:
js8groups = @ALLCAL,@HB,@TC2

Nodes with Urgent board permissions: []
2024-12-14 14:46:18 - INFO - TC²-BBS is running on serial interface...
Database schema initialized.
2024-12-14 14:46:18 - JS8Call - INFO - Database tables created or verified.
2024-12-14 14:46:18 - JS8Call - INFO - Connecting to ('192.168.0.114', 2442)
2024-12-14 14:47:43 - JS8Call - INFO - Received JS8Call message: OE1TMW: to OE1TMW/P - SNR +05 ♢
2024-12-14 14:48:28 - JS8Call - INFO - Received JS8Call message: OE1TMW: to OE1TMW/P - SNR +03 ♢
2024-12-14 14:54:28 - JS8Call - INFO - Received JS8Call message: 9A3SWO: to @HB - HEARTBEAT ♢
Traceback (most recent call last):
  File "/home/pi/TC2-BBS-mesh/server.py", line 93, in <module>
    main()
  File "/home/pi/TC2-BBS-mesh/server.py", line 80, in main
    js8call_client.connect()
  File "/home/pi/TC2-BBS-mesh/js8call_integration.py", line 206, in connect
    self.process(message)
  File "/home/pi/TC2-BBS-mesh/js8call_integration.py", line 167, in process
    self.insert_message('groups', sender, receiver, msg)
  File "/home/pi/TC2-BBS-mesh/js8call_integration.py", line 129, in insert_message
    ''', (sender, receiver_or_group, message))
NameError: name 'receiver_or_group' is not defined                                                   (TC2-BBS stopped)
(venv) pi@bullseye15:~/TC2-BBS-mesh $ python server.py

Have I missed something, any tip is appreciated

Thomas

TheCommsChannel, 2024-12-21:
I'll have to do some testing and see what's going on. Does this only happen for @hb?

myerika, 2024-12-21:
Hi

Same problem with all @groups. And with direct messages when store_messages
= True is set.

kc7hes, 2025-03-06:
From one of my Smarter than Me buddies:

For the TC2-BBS-mesh project:
in the js8call_integration.py file, change the following lines from/to:
line 129: Change:
                ''', (sender, receiver_or_group, message))
to
                ''', (sender, recipient, message))

myerika, 2025-03-06:
Works fine now
Thanks
Thomas