TheCommsChannel/TC2-BBS-mesh #114: No serial port detected on tcp.

Spogster, 2025-02-16:
I have just updated to latest version now I have have the following.

  File "/home/mesh/bbs/config_init.py", line 170, in get_interface

    raise ValueError("No serial ports detected.")

ValueError: No serial ports detected.

My config file says 

type = tcp
hostname = 192.168.1.16

I have checked that I can access from the server.

I am running meshtastic on Raspberry PI and docker for TC2BBS with version v1.0.47

NerdsCorp, 2025-03-22:
im getting the same thing,

NerdsCorp, 2025-03-22:
check your config file the location is diffrent on the location

https://github.com/TheCommsChannel/TC2-BBS-mesh/blob/295fb35c92e376367b05a6429c481191baa23d17/docker/docker-compose.yaml