TheCommsChannel/TC2-BBS-mesh #117: Docker Volume

walmer-bolfet, 2025-02-17:
When I try to launch a docker container with your docker image, it creates its own anonymous docker volume instead of the bind mount (and volume mount) I created and tried to use with the launch container command.  I don’t have an expansion understanding of docker, but I have recreated this condition consistently. 

The container works as expected, and I can edit the config.ini file in the anonymous docker volume to get changes upon restarting the container.  

I appreciate your good work and wonder if you might understand this behavior. 

Thank you.

TheCommsChannel, 2025-02-18:
Not sure what's going on there. Someone else created the docker image

closed 2025-02-18 by TheCommsChannel

walmer-bolfet, 2025-02-18:
I see my mistake:

https://hub.docker.com/r/thealhu/tc2-bbs-mesh

walmer-bolfet, 2025-02-18:
I am still learning.  Perhaps this will be valuable to other:

The Dockerfile provided contains a `VOLUME` instruction, which is the reason bind mount is not working as expected. 

- 'VOLUME /home/mesh/bbs/config`: 

This line tells Docker that the `/home/mesh/bbs/config` directory inside the container should be a Docker-managed volume. When Docker encounters this instruction, it creates an anonymous volume automatically. Regardless of whether one tries to use a bind mount, Docker will create an anonymous volume for `/home/mesh/bbs/config` before the bind mount is applied. 

In summary: The `VOLUME` instruction in the Dockerfile overrides attempts to specify a bind mount or named volume.