TheCommsChannel/TC2-BBS-mesh #118: Post view delivers out of order or not at all

bmidgley, 2025-02-20:
[image]

Tc2 sends these using bbs7 (my tbeam) in order but they always arrive like this in the end. There's a several second delay between the menu and the post.

If I reply before the post arrives then I never get it.

This seems like meshtastic issues, but maybe there is a workaround. Can you get a confirmation the post delivery succeeded before sending the menu?

shortwavesurfer2009, 2025-02-26:
Possibly could add some logic to wait for delivery confirmation before sending the next message, although I think that would break if the delivery confirmation never arrived, and so maybe a timer would be better where it sends the message waits like five seconds and then sends the next message.

AdvJosh, 2025-03-05:
> Possibly could add some logic to wait for delivery confirmation before sending the next message, although I think that would break if the delivery confirmation never arrived, and so maybe a timer would be better where it sends the message waits like five seconds and then sends the next message.

I've been toying with that idea; I think there are a few things to do that would help:
1. Slow down messages; the rapid fire nature of how it currently works can cause issues. Ideally, the delay between messages should be dependent on the modem preset; Short-Turbo can fire off messages far quicker that Long-Fast. Alternatively, just make it a config value and users can tweak it to their liking.
2. Try to wait for ACKs before sending the next message. To avoid the issues that you mentioned, I would put a maximum time limit on how long to wait before proceeding. Do you think it would be helpful to have it retry sending the message a second time if no ACK is received?
3. A channel utilization check - I am still looking into how to get this information through the python API and weather it would only be updated once a minute like on the apps. The general idea being to wait for channel utilization to be under 15-20% to avoid TX errors and packet collisions
4. Flatter menu and/or more quick commands - the more we reduce the amount of data the bbs sends, the less impact it will have on the overall mesh

I think these solutions would help resolve some of the issues. I will look into these options, and more then like make PRs one-by-one as I get each one working. I'd be curious to hear your thoughts about it!

TheCommsChannel, 2025-03-05:
@AdvJosh Appreciate all of your help with this project! I'll have a look at your current PR as soon as I get some free time and any upcoming ones your put in.

varna9000, 2025-05-26:
I have similar issue. After the first message I receiving the BBS initial menu ok. However after choosing an item from that menu, client node waits for ACK from the BBS node, while the BBS fires immediately the menu msg (according to the debug log). The client is still waiting the first ACK and misses the BBS menu msg.